john30/ebusd-configuration #500: ebusd hwc mctype 9 unkown

garageboxnv, 2025-04-20:
Using hassio mqtt integration gives me following error:
```
Ignoring invalid option received on topic 'ebusd/hwc/Mode', got '9', allowed: inactive, mixer, fixed, hwc, returnincr, pool, circulation
```
I have an vaillant aguaflow exclusive, I expect mode 9 might indicate that the mctype is a "buffercircuit"

mqtt hassio state topic
```
unique_id: ebusd_hwc_Mode_mctype
device:
  identifiers: ebusd_hwc
  manufacturer: ebusd.eu
  via_device: ebusd
  sw_version: '24.1'
  suggested_area: Heating
  name: ebusd hwc
value_template: '{{value_json["mctype"].value}}'
state_topic: ebusd/hwc/Mode
device_class: enum
options:
  - inactive
  - mixer
  - fixed
  - hwc
  - returnincr
  - pool
  - circulation
name: Mode mctype
```
value received
```
Received 08:31:48

    QoS: 0, Retained
    Payload:

    temp:
      value: 47
    hwcmode:
      value: auto
    hwcmode_1:
      value: disabled
    mctype:
      value: 9
    hex:
      value: '00'
    daynight:
      value: night
```

prehor, 2025-11-29:
I get the same error. My suspicion is that the mode means Eco HWC. 

@garageboxnv are you also using this mode for hot water heating?

garageboxnv, 2025-11-29:
I believe you refer to the circulation pump, although installed I have disabled it, mode off.
In future want tobdo some smart control of the pump but not yet figured out how

I think this mctype refers to the 'boilerpump' as mentioned in manual, which feeds the heat exchanger of the drinking water station