john30/ebusd-configuration #505: Ignoring invalid option received on topic 'ebusd/hmu/SetMode'

kecksk0enig, 2025-05-03:
Seeing this eror in homeassistant log:

` WARNING (MainThread) [homeassistant.components.mqtt.sensor] Ignoring invalid option received on topic 'ebusd/hmu/SetMode', got '3', allowed: auto, off, water
`

I found this issue on ebusd github page:
https://github.com/john30/ebusd/issues/1350

however no further comment if this issue is already beeing worked on in ebusd-config

MarcelT87, 2026-10-05:
I have a new reproducible data point for the `SetMode` enum issue, this time with value `23`.

System here:

- HMU00 at `0x08`
- SW `0902`
- HW `5103`
- gateway-originated traffic from `0xf1`

In the current database I found **24 natural `f1 -> 08 B510/00` telegrams** with the same payload shape:

```text
0b 00 17 00 00 00 00 00 00 00 00 00
```

The response is consistently:

```text
01 00
```

Decoded with the current definition this becomes effectively:

```text
hcmode = 23
flowtempdesired = 0.0
...
```

What is interesting is the timing.

From 2026-09-22 through 2026-10-05, one such telegram appears almost exactly once per day around:

```text
05:26:15 local time
```

The normal controller (`QQ=10`) continues to send its normal `SetMode` telegrams and resumes shortly afterwards.

For the observed `f1` frames, the delay until the next normal `QQ=10` SetMode is roughly:

```text
min    0.476 s
median 5.861 s
max    9.336 s
```

There is also prior independent evidence of the same symptom:

- `john30/ebusd#1636` reported `hmu/SetMode = 23`.
- John already noted there that this looks like a **definition issue in `hcmode_inc.tsp`**, not an ebusd core issue.
- Other Vaillant installations with VR921/VR940f have also reported recurring gateway-originated `SetMode ... = 23`.

One detail may be important for the actual fix.

A normal controller SetMode master payload here has length `NN=09`, for example:

```text
09 00 00 3e ff ff ff 06 00 00
```

whereas the `f1`/`23` telegram uses:

```text
0b 00 17 00 00 00 00 00 00 00 00 00
```

So the gateway-originated frame is **two payload bytes longer** than the normal SetMode variant.

Because of that, I am not sure whether this is simply:

```text
Values_hcmode: add 23
```

or whether the `f1 B510/00` telegram is actually a distinct/extended gateway-specific SetMode variant that only shares the same prefix.

I would therefore prefer not to guess an enum label yet.

Could someone confirm whether this should be handled as:

1. a missing `hcmode` value, or
2. a separate/extended `B510/00` definition for the gateway-originated frame?

I can provide more raw examples or timing/context from the same installation if useful.

MarcelT87, 2026-10-06:
I did the focused read-only context check mentioned above, and this seems to resolve the local "hcmode=23" ambiguity quite cleanly.

The gateway frame in question is:

f1 -> 08 B510
0b 00 17 00 00 00 00 00 00 00 00 00

I compared every natural occurrence against the neighboring B531 LPC traffic in the retained database.

The result is a complete 1:1 temporal coupling:

B510 /0b0017 frames:                 24
HMU+VWZIO B531 /0301ffff pairs:      24

inactive pair -> B510 within 2 s:    24 / 24

delay:
min     0.184 s
median  0.185 s
max     0.641 s

The controlled EEBUS LPC-disable experiment is one member of exactly this same population.

The automatically recurring events around 05:26 also reproduce the same ordering:

HMU   B531 /0301ffff
VWZIO B531 /0301ffff
        ↓
HMU   B510 /0b0017...
        ↓
CTLV2 B531 /020301   Gateway-LPC signal absent

For the controlled transitions, "/020301" follows almost immediately. During the daily refresh events it can follow about 10.8 s later, but the "ffff/ffff -> B510 /0b0017" relationship remains the same.

So on this installation I would no longer interpret byte "0x17" as evidence for a missing global "hcmode=23" enum value.

The stronger interpretation is that this longer gateway-specific B510 frame is an LPC-inactive transition/companion frame, and the current SetMode definition is simply decoding its "0x17" byte through the normal SetMode layout.

I would still leave the exact official B510 message/field name open, and this is currently a strong local identification rather than a portable cross-install definition.

But I think this makes adding "23" to the global "hcmode" enum the wrong fix.