john30/ebusd-configuration #506: Hc1RoomTempMod missing in new configuration

stingone, 2025-05-06:
I finally migrated from version 23 to 25.
In the old configuration i had the topic mode_state_topic: "ebusd/ctlv2/Hc1RoomTempMod" used for my thermostat.

What is the new topic for this?

Old:
r;w,,Hc1RoomTempMod,,,,,1500,,,rcmode2,,,room temperature modulation of Hc1

stingone, 2025-05-08:
I found al corresponding mappings in the "new" CSV file. only one that is missing is Z1roomhumidity. 
Is there an easy way to add this value in a easy way local to the CSV file?

r, z1RoomHumidity, B524, 020003002800, EXP, %

Tried to add a row. but as soon i reload the CVS it gives me an error.

JonesGW-MGD, 2025-05-19:
What error do you get?

Have you tried

```r,ctlv2,z1RoomHumidity,z1RoomHumidity,31,15,B524,020003002800,value,,IGN:4,,,,value,,EXP,,%,z1 Room Humidity```

This works for me albeit earlier release.

(presumably your spaces after commas are not intended?)