john30/ebusd-configuration #511: A lot of datasets not found

overclocka, 2025-05-30:
First of all, I would like to thank the developers very much for this amazing project!

Unfortunately, I currently have the problem that no values are loaded for many parameters and I wanted to be sure that I have not made a fundamental mistake

I have a C6 shield adapter and:

- An aroTHERM pure VWL 45/7.2 AS 230V S3 heat pump
- A VRC720 control panel
- A uniTOWER VWL 108/7.7 IS C2

I think I use the latest version:
`ebusd -V
ebusd 25.1.25.1
`
Some entries tell me that there are newer CSV files:
`version: ebusd 25.1.25.1
update check: OK, device firmware v5-c6 1[5511] up to date, broadcast.csv: different version available, memory.csv: different version available, vaillant/08.hmu.csv: different version available, vaillant/broadcast.csv: different version available, vaillant/ec.sol.sc.csv: different version available, vaillant/general.csv: different version available, vaillant/scan.csv: different version available
device: 192.168.1.30:9999, TCP, enhanced, firmware 1.1[5511].1[5511]
signal: acquired
symbol rate: 22
max symbol rate: 160
min arbitration micros: 0
max arbitration micros: 81
min symbol latency: 4
max symbol latency: 7
scan: finished
reconnects: 0
masters: 4
messages: 757
conditional: 0
poll: 1
update: 29
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0129;HW=2204", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV3;SW=0709;HW=3704", loaded "vaillant/15.ctlv3.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0102;HW=0504", loaded "vaillant/76.vwzio.csv"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0709;HW=3704", loaded "vaillant/ec.sol.sc.csv"
`
Some values are determined here, but not too many at present:
`ebusctl find -d
Broadcast Outsidetemp = 20.098
Broadcast Vdatetime = 11:43:36;30.05.2025
hmu DateTime = valid;11:43:38;30.05.2025;20.148
hmu SetMode = auto;20.0;-;-;1;1;1;0;0;1
hmu Status01 = 21.0;21.0;20.098;-;45.5;off
hmu StatusCirPump = on
sc YieldThisYear =  (ERR: invalid position for 31ecb509030d3f00 / 00)
scan.08  = Vaillant;HMU00;0129;2204
Scan.08 Id = 21;24;14;0010038163;3130;325063;N0
scan.15  = Vaillant;CTLV3;0709;3704
Scan.15 Id = 21;24;09;0020260913;0953;011170;N0
scan.76  = Vaillant;VWZIO;0102;0504
Scan.76 Id = 21;24;14;0010038163;3130;325063;N0
scan.ec  = Vaillant;SOL00;0709;3704
Scan.ec Id = 21;24;09;0020260913;0953;011170;N0
`

At first I thought that it might take a few minutes for other values to be loaded. Unfortunately, this was not the case. Below I show the current log file after I have emptied it and restarted EBUSD:
`systemctl start ebusd && tail -f /var/log/ebusd.log
2025-05-30 11:46:09.117 [device notice] transport opened
2025-05-30 11:46:09.118 [main notice] ebusd 25.1.25.1 started with full scan on device: 192.168.1.30:9999, TCP, enhanced
2025-05-30 11:46:09.119 [bus notice] bus started with own address 31/36
2025-05-30 11:46:09.120 [device notice] reset, supports info
2025-05-30 11:46:09.122 [device notice] extra info: firmware 1.1[5511].1[5511], jumpers 0x0b
2025-05-30 11:46:09.133 [bus notice] signal acquired
2025-05-30 11:46:15.400 [bus notice] new master 10, master count 2
[... 157 lines cut ...]
2025-05-30 11:53:52.796 [update notice] received read hmu Status01 QQ=10: 21.0;21.0;20.199;-;45.5;off
2025-05-30 11:53:55.039 [update notice] received unknown MS cmd: 1076b5110101 / 092a2a3314ff5b0000ff
2025-05-30 11:53:57.282 [update notice] received unknown MS cmd: 1008b5110100 / 0951010f006408000000
2025-05-30 11:53:59.575 [update notice] received unknown MS cmd: 1076b512030f0101 / 07d8020053010fff
2025-05-30 11:54:00.924 [update notice] received update-write hmu SetMode QQ=10: auto;20.0;-;-;1;1;1;0;0;1
2025-05-30 11:54:07.089 [update notice] sent scan-read scan.76  QQ=31: Vaillant;VWZIO;0102;0504
2025-05-30 11:54:07.089 [bus notice] scan 76: ;Vaillant;VWZIO;0102;0504
2025-05-30 11:54:07.234 [update notice] received unknown MS cmd: 1076b51009000000ffffff010000 / 0101
2025-05-30 11:54:09.426 [update notice] received update-read Broadcast Outsidetemp QQ=10: 20.250
2025-05-30 11:54:11.672 [update notice] received read hmu Status01 QQ=10: 21.0;21.0;20.250;-;45.5;off
2025-05-30 11:54:13.914 [update notice] received unknown MS cmd: 1076b5110101 / 092a2a4014ff5b0000ff
2025-05-30 11:54:16.153 [update notice] received unknown MS cmd: 1076b512030f0101 / 07d9020053010fff
2025-05-30 11:54:18.390 [update notice] received update-write hmu SetMode QQ=10: auto;20.0;-;-;1;1;1;0;0;1
2025-05-30 11:54:20.624 [update notice] received unknown MS cmd: 1076b51009000000ffffff010000 / 0101
2025-05-30 11:54:22.866 [update notice] received read hmu Status01 QQ=10: 21.0;21.0;20.250;-;45.5;off
2025-05-30 11:54:25.107 [update notice] received unknown MS cmd: 1076b5110101 / 092a2a4014ff5b0000ff
`

Now to my question: Am I right in assuming that my system is not covered by the configuration files and that I am therefore not getting any essential data here?

As for the heat pump, I'm relatively sure that there aren't many configurations for it yet, as a search on the model didn't turn up much for me. However, I'm surprised that I didn't get any values for the SensoComfort VRC720. Does anyone have any tips for me?

If there is no other way, I will probably have to delve deeper into the matter and try to decode things myself. However, I don't get any more specific messages via the EBUSD when I make changes to the VRC720 (possibly because changes to the settings are not sent instantly here), which makes interpretation difficult. Does anyone have any tips on how to proceed?

Thank you very much!

jamski68, 2025-06-16:
same questions from me too...

CommanderROR9, 2025-08-21:
Unfortunately the whole ebusd project remains quite experimental.
My Arotherm Plus 55/6 works reasonably well, but entities that were available for Months sometimes just vanish, never to be seen again, without any discerible reason.
For example, I have the "Energy Integral" value for a long time, but recently it just vanished and I have not been able to get it back. This means any Automation or Dashboard build with that Entity is now just dead.